althafvly/AmlogicKitchen #3: Windows repack makes read only partition.

dister520, 2022-05-22:
1. take an image and unpack level 1.

Level1 repack shows no issues

 2. going into adb shell user is allowed to su and get # 
 3. Unpack 2 and repack both 2 and 1
 4. Try step 2 and user receives su not permitted. Image is read only.
 
 much thanks

althafvly, 2022-05-23:
Are you trying to get root access?. If so, this tool does not add super user access into a rom. You need manually add it.

dister520, 2022-05-23:
Attached is the screenshot of the rom from vendor which allows su from shell. It appears that after I pack with 2 the item is changed. Wanting shell to have access to su like the vendor rom.
[image]

dister520, 2022-05-24:
I believe this is an issue related to unpack and not sure if fs_context and or permissions are changed by the imgextractor?

althafvly, 2022-05-24:
Just unpacking and packing won't remove su, if you already have. Link me your firmware, i can check

dister520, 2022-05-25:
thanks emailed link to the gmail account in profile.

althafvly, 2022-05-26:
I just repacked your firmware. it seems su is present here. hasnt removed

dister520, 2022-05-26:
Great please see the permission of su in system/xbin/su. 
The permissions will have changed and be different than the above screenshot.

dister520, 2022-05-26:
Before packing the system would allow su without root. After repacking su would require adb-root.

althafvly, 2022-05-26:
- You can add/change permission in level2/config/*_fs_config for each files and folder manually.

- adb root is only for development builds, which means your build is userdebug/eng.

**Keep in mind that this is dev tool, try everything at your own risk.**

closed 2022-05-26 by althafvly

dister520, 2022-05-27:
Correct it’s a debug build. And not sure why when I repack fs file it does not function as a debug build.

   
   -