althafvly/AmlogicKitchen #4: Windows pack block size is 0 during repack and not 4k

dister520, 2022-05-22:
Repack of .img block of 4K Is not seen.
Linux appears to be ok but super. Partition is not created on Linux due to type field and super_type.txt missing.

althafvly, 2022-05-23:
Can you explain in bit more details?. which script, platform and firmware are you using ?

dister520, 2022-05-23:
Apologies breaking up the comments. 
- Linux unpack has errors with the python. 
On windows with batch file packing shows an abstract like this which is different than linux, which shows (4k) blocks long.
Windows snippet//////////
Resizing the filesystem on level2\odm_a.img to 193 (4k) blocks.
The filesystem on level2\odm_a.img is now 193 (0k) blocks long.
/////////////////////////////////////////////
Log below for complete log and issue of sparse / magic seen as well. 
.....................
Amlogic Kitchen
.....................
Select level 1,2 or 3: 2
Packing odm_ext_a
loaded 58 fs_config entries
Creating filesystem with parameters:
    Size: 16777216
    Block size: 4096
    Blocks per group: 32768
    Inodes per group: 1024
    Inode size: 256
    Journal blocks: 0
    Label: odm_ext_a
    Blocks: 4096
    Block groups: 1
    Reserved block group size: 7
Created filesystem with 37/1024 inodes and 1721/4096 blocks
Packing system_a
loaded 2880 fs_config entries
Creating filesystem with parameters:
    Size: 1150390272
    Block size: 4096
    Blocks per group: 32768
    Inodes per group: 7808
    Inode size: 256
    Journal blocks: 0
    Label: system_a
    Blocks: 280857
    Block groups: 9
    Reserved block group size: 71
Created filesystem with 2888/70272 inodes and 264604/280857 blocks
resize2fs 1.42.12 (29-Aug-2014)
Resizing the filesystem on level2\system_a.img to 264636 (4k) blocks.
The filesystem on level2\system_a.img is now 264636 (0k) blocks long.

Packing system_ext_a
loaded 82 fs_config entries
Creating filesystem with parameters:
    Size: 123547648
    Block size: 4096
    Blocks per group: 32768
    Inodes per group: 7552
    Inode size: 256
    Journal blocks: 0
    Label: system_ext_a
    Blocks: 30163
    Block groups: 1
    Reserved block group size: 7
Created filesystem with 90/7552 inodes and 28429/30163 blocks
resize2fs 1.42.12 (29-Aug-2014)
Resizing the filesystem on level2\system_ext_a.img to 28432 (4k) blocks.
The filesystem on level2\system_ext_a.img is now 28432 (0k) blocks long.

Packing vendor_a
loaded 1034 fs_config entries
Creating filesystem with parameters:
    Size: 296427520
    Block size: 4096
    Blocks per group: 32768
    Inodes per group: 6032
    Inode size: 256
    Journal blocks: 0
    Label: vendor_a
    Blocks: 72370
    Block groups: 3
    Reserved block group size: 23
Created filesystem with 1033/18096 inodes and 68003/72370 blocks
resize2fs 1.42.12 (29-Aug-2014)
Resizing the filesystem on level2\vendor_a.img to 68011 (4k) blocks.
The filesystem on level2\vendor_a.img is now 68011 (0k) blocks long.

Packing product_a
loaded 129 fs_config entries
Creating filesystem with parameters:
    Size: 145985536
    Block size: 4096
    Blocks per group: 32768
    Inodes per group: 4464
    Inode size: 256
    Journal blocks: 0
    Label: product_a
    Blocks: 35641
    Block groups: 2
    Reserved block group size: 15
Created filesystem with 137/8928 inodes and 33633/35641 blocks
resize2fs 1.42.12 (29-Aug-2014)
Resizing the filesystem on level2\product_a.img to 33637 (4k) blocks.
The filesystem on level2\product_a.img is now 33637 (0k) blocks long.

Packing odm_a
loaded 21 fs_config entries
Creating filesystem with parameters:
    Size: 1048576
    Block size: 4096
    Blocks per group: 32768
    Inodes per group: 64
    Inode size: 256
    Journal blocks: 0
    Label: odm_a
    Blocks: 256
    Block groups: 1
    Reserved block group size: 7
Created filesystem with 29/64 inodes and 193/256 blocks
resize2fs 1.42.12 (29-Aug-2014)
Resizing the filesystem on level2\odm_a.img to 193 (4k) blocks.
The filesystem on level2\odm_a.img is now 193 (0k) blocks long.

Packing super
lpmake I 05-23 14:46:51   371 17192 builder.cpp:1093] [liblp]Partition system_a will resize from 0 bytes to 1083949056 bytes
lpmake I 05-23 14:46:51   371 17192 builder.cpp:1093] [liblp]Partition system_ext_a will resize from 0 bytes to 116457472 bytes
lpmake I 05-23 14:46:51   371 17192 builder.cpp:1093] [liblp]Partition vendor_a will resize from 0 bytes to 278573056 bytes
lpmake I 05-23 14:46:51   371 17192 builder.cpp:1093] [liblp]Partition product_a will resize from 0 bytes to 137777152 bytes
lpmake I 05-23 14:46:51   371 17192 builder.cpp:1093] [liblp]Partition odm_a will resize from 0 bytes to 790528 bytes
Invalid sparse file format at header magic
Invalid sparse file format at header magic
Invalid sparse file format at header magic
Invalid sparse file format at header magic
Invalid sparse file format at header magic

althafvly, 2022-05-24:
Can you tell what's wrong in this?. It's packing super image. Super image contains odm,system,system_ext and vendor partition.

althafvly, 2022-05-26:
This is done in both linux and windows version of this script. warning was hidden in windows.

closed 2022-05-26 by althafvly

dister520, 2022-05-26:
The warning in Linux says it repacked with 4K block and the above log shows warning it repacked in 0k block. There would be a discrepancy between Linux fs and windows fs.

althafvly, 2022-05-26:
That warning is present in both linux and windows. I just reverted it old state to avoid anymore confusion. You can use the latest changes and check