althafvly/AmlogicKitchen #5: Latest Commit e3d9e93f99a0239106e69ac60cf2a9963ee15073 error unpacking level 3

subjec2change, 2022-06-17:
I have this error when unpacking aml_upgrade_package.img using the windows version:

.....................
Amlogic Kitchen
.....................
Select level 1,2 or 3: 3
        1 file(s) copied.
Android Image Kitchen - UnpackImg Script
by osm0sis @ xda-developers

Supplied image: recovery.img

Setting up work folders . . .

Image type: AOSP

Signature with "AVBv2" type detected.

Splitting image to "split_img/" . . .

ANDROID! magic found at: 0
BOARD_KERNEL_CMDLINE androidboot.dtbo_idx=0 --cmdline otg_device=1 buildvariant=user
BOARD_KERNEL_BASE 0x01078000
BOARD_NAME
BOARD_PAGE_SIZE 2048
BOARD_HASH_TYPE sha1
BOARD_KERNEL_OFFSET 0x00008000
BOARD_RAMDISK_OFFSET 0xfff88000
BOARD_SECOND_OFFSET 0xffe88000
BOARD_TAGS_OFFSET 0xfef88100
BOARD_OS_VERSION 9.0.0
BOARD_OS_PATCH_LEVEL 2021-12
BOARD_HEADER_VERSION 1
BOARD_RECOVERY_DTBO_SIZE 438
BOARD_RECOVERY_DTBO_OFFSET 16318464
BOARD_HEADER_SIZE 1648

Unpacking ramdisk to "ramdisk/" . . .

Compression used: gzip
20003 blocks

Done

Press any key to continue . . .
        1 dir(s) moved.
        1 dir(s) moved.
        1 file(s) copied.
Android Image Kitchen - UnpackImg Script
by osm0sis @ xda-developers

Supplied image: boot.img

Setting up work folders . . .

Image type: AOSP

Signature with "AVBv2" type detected.

Splitting image to "split_img/" . . .

ANDROID! magic found at: 0
BOARD_KERNEL_CMDLINE androidboot.dtbo_idx=0 --cmdline otg_device=1 buildvariant=user
BOARD_KERNEL_BASE 0x01078000
BOARD_NAME
BOARD_PAGE_SIZE 2048
BOARD_HASH_TYPE sha1
BOARD_KERNEL_OFFSET 0x00008000
BOARD_RAMDISK_OFFSET 0xfff88000
BOARD_SECOND_OFFSET 0xffe88000
BOARD_TAGS_OFFSET 0xfef88100
BOARD_OS_VERSION 9.0.0
BOARD_OS_PATCH_LEVEL 2021-12
BOARD_HEADER_VERSION 1
BOARD_HEADER_SIZE 1648

Warning: No ramdisk found to be unpacked

Done

Press any key to continue . . .
        1 dir(s) moved.
        1 dir(s) moved.
dbg:res-img ver is 0x2
dbg:item level3\logo/upgrade_error
dbg:item level3\logo/upgrade_success
dbg:item level3\logo/upgrade_bar
dbg:item level3\logo/bootup
dbg:item level3\logo/upgrade_logo
dbg:item level3\logo/upgrade_unfocus
dbg:item level3\logo/upgrade_upgrading
dbg:item level3\logo/upgrade_fail
FATAL ERROR: Blob has incorrect magic number
ERROR: level1\_aml_dtb.PARTITION
Can not open the file as archive

DTB Version: 2 entries: 2
Found header: g12a-u212-2g
         offset: 2048 size: 78496
Found header: g12a-u212-4g
         offset: 81920 size: 78524

Done.
Press any key to continue . . .

I'm worried that even if I don't make any changes the img will not repack back to its same size and be bootable.
Can you advise on this error?

althafvly, 2022-06-20:
You can ignore that error, if the dts is being extracted in level3. I already mentioned this in the readme notes

subjec2change, 2022-06-21:
oops now i feel embarrassed...

 Quick question if I unpack level 2 and then repack down to level 1 and then make that a IMG file.. when I flash it the device will bootloop even if I don't change anything in the files of level 2. I have been banging my head on changing a rom to have usb debugging OOTB for logging. for reference its a S905X2 chip.

althafvly, 2022-06-21:
> oops now i feel embarrassed...
>
> Quick question if I unpack level 2 and then repack down to level 1 and then make that a IMG file.. when I flash it the device will bootloop even if I don't change [...]

link me the firmware image. i can check

closed 2022-07-04 by althafvly