althafvly/AmlogicKitchen #6: error when unpacking level 2 using python

karnadii, 2022-07-22:
i tried to unpack to level 2 using linux and windows. windows fine using vortex, got error when using python
```bash
.....................
Amlogic Kitchen
.....................
.....................
Select level 1,2,3 or q/Q to exit:
2
Deleting existing level2
Extracting system
Invalid sparse file format at header magic
Failed to read sparse file
.....Extraction from system.raw.img to system
Traceback (most recent call last):
  File "bin/common/imgextractor.py", line 655, in <module>
    Extractor().main(sys.argv[1], (sys.argv[2] + os.sep + os.path.basename(sys.argv[1]).split('.')[0]))
  File "bin/common/imgextractor.py", line 650, in main
    self.__ext4extractor()
  File "bin/common/imgextractor.py", line 473, in __ext4extractor
    root = ext4.Volume(file).root
  File "/home/karnadi/AmlogicKitchen/bin/common/ext4.py", line 479, in __init__
    self.superblock = self.read_struct(ext4_superblock, 0x400)
  File "/home/karnadi/AmlogicKitchen/bin/common/ext4.py", line 529, in read_struct
    return structure._from_buffer_copy(raw, platform64=platform64 if platform64 != None else self.platform64)
  File "/home/karnadi/AmlogicKitchen/bin/common/ext4.py", line 320, in _from_buffer_copy
    struct = ext4_superblock.from_buffer_copy(raw)
ValueError: Buffer size too small (0 instead of at least 1024 bytes)
Extracting vendor
Invalid sparse file format at header magic
Failed to read sparse file
.....Extraction from vendor.raw.img to vendor
Traceback (most recent call last):
  File "bin/common/imgextractor.py", line 655, in <module>
    Extractor().main(sys.argv[1], (sys.argv[2] + os.sep + os.path.basename(sys.argv[1]).split('.')[0]))
  File "bin/common/imgextractor.py", line 650, in main
    self.__ext4extractor()
  File "bin/common/imgextractor.py", line 473, in __ext4extractor
    root = ext4.Volume(file).root
  File "/home/karnadi/AmlogicKitchen/bin/common/ext4.py", line 479, in __init__
    self.superblock = self.read_struct(ext4_superblock, 0x400)
  File "/home/karnadi/AmlogicKitchen/bin/common/ext4.py", line 529, in read_struct
    return structure._from_buffer_copy(raw, platform64=platform64 if platform64 != None else self.platform64)
  File "/home/karnadi/AmlogicKitchen/bin/common/ext4.py", line 320, in _from_buffer_copy
    struct = ext4_superblock.from_buffer_copy(raw)
ValueError: Buffer size too small (0 instead of at least 1024 bytes)
Extracting product
Invalid sparse file format at header magic
Failed to read sparse file
.....Extraction from product.raw.img to product
Traceback (most recent call last):
  File "bin/common/imgextractor.py", line 655, in <module>
    Extractor().main(sys.argv[1], (sys.argv[2] + os.sep + os.path.basename(sys.argv[1]).split('.')[0]))
  File "bin/common/imgextractor.py", line 650, in main
    self.__ext4extractor()
  File "bin/common/imgextractor.py", line 473, in __ext4extractor
    root = ext4.Volume(file).root
  File "/home/karnadi/AmlogicKitchen/bin/common/ext4.py", line 479, in __init__
    self.superblock = self.read_struct(ext4_superblock, 0x400)
  File "/home/karnadi/AmlogicKitchen/bin/common/ext4.py", line 529, in read_struct
    return structure._from_buffer_copy(raw, platform64=platform64 if platform64 != None else self.platform64)
  File "/home/karnadi/AmlogicKitchen/bin/common/ext4.py", line 320, in _from_buffer_copy
    struct = ext4_superblock.from_buffer_copy(raw)
ValueError: Buffer size too small (0 instead of at least 1024 bytes)
Extracting odm
.....Extraction from odm.raw.img to odm
.....Done! All extraction in odm
Done.
```

althafvly, 2022-07-22:
Link me your firmware, so i can check

karnadii, 2022-07-22:
here is the firmware https://androidfilehost.com/?fid=14943124697586373056

althafvly, 2022-07-22:
I have pushed some changes now. it should unpack every partition except system partition in this firmware for linux. Reason linux can't unpack system partition is because this image contains something that breaks linux tool. you might need to cleanup unwanted files from system to make it work with linux.

althafvly, 2022-07-22:
There's file starts with name "Firmware By "* in system partition. That file name contains spaces, which we dont really use in linux or android in filesystem. Remove it and pack, it should work on linux now.
I have modified that file for you, download it from https://downloads.althafvly.in/s/pQniXjdyejxkJsm or you can try by removing that file.

closed 2022-07-22 by althafvly

karnadii, 2022-07-22:
thanks, but I have another question, is it normal when the file size shrinking when using your tool? I notice up to 200-300Mb size shrinking in the final packed version compared normal version

althafvly, 2022-07-22:
Image size change should be fine. Script sparse partition images instead of raw ext4 image to reduce size. If you have doubts if something removed check if there's any such difference in compressed image with zip or 7z.