althafvly/AmlogicKitchen #8: FATAL ERROR: Blob has incorrect magic number | ERROR: level1\_aml_dtb.PARTITION

ShelleyLake, 2022-08-08:
Hi @xKern,
I have a problem with a "boot.img" image of a device labeled "Angel" (on Alixpress and similar sites it is called "Angel V8").
When I run the "unpacke" command and then choose level three, I get the above error message.
Can you give me some more information?

Thank you very much.

J.

```
Supplied image: boot.img

Setting up work folders . . .

Image type: AOSP

Splitting image to "split_img/" . . .

ANDROID! magic found at: 0
BOARD_KERNEL_CMDLINE
BOARD_KERNEL_BASE 0x10000000
BOARD_NAME
BOARD_PAGE_SIZE 2048
BOARD_HASH_TYPE sha1
BOARD_KERNEL_OFFSET 0x00008000
BOARD_RAMDISK_OFFSET 0x01000000
BOARD_SECOND_OFFSET 0x00f00000
BOARD_TAGS_OFFSET 0x00000100
BOARD_HEADER_VERSION 0

Unpacking ramdisk to "ramdisk/" . . .

Compression used: gzip
10461 blocks

Done

dbg:res-img ver is 0x2
dbg:item level3\logo/bootup
dbg:item level3\logo/upgrade_error
dbg:item level3\logo/upgrade_fail
dbg:item level3\logo/bootup.bmp
dbg:item level3\logo/upgrade_bar
dbg:item level3\logo/upgrade_success
dbg:item level3\logo/upgrade_unfocus
dbg:item level3\logo/upgrade_upgrading
dbg:item level3\logo/upgrade_logo
FATAL ERROR: Blob has incorrect magic number
ERROR: level1\_aml_dtb.P
[Image.zip](https://github.com/xKern/AmlogicKitchen/files/9280881/Image.zip)
ARTITION
Can not open the file as archive

DTB Version: 2 entries: 2
Found header: m8b-m201-1G
         offset: 2048 size: 23139
Found header: m8b-m201C-512M
         offset: 26624 size: 22031

Done.
```

althafvly, 2022-08-08:
It's already noted in readme.

closed 2022-08-08 by althafvly

ShelleyLake, 2022-08-08:
Thanks ... I'll just ignore that error message.