althafvly/AmlogicKitchen #10: Thank you for the tool you provided

jikeku, 2022-09-09:
Thank you for the tool you provided. I used your tool to unzip my android 11 package, but it was not successful. Can you support it?
Download：https://mega.nz/file/jiY0wLJD#4EhVCEM0ykJaCF5L1oUzSMLDNqIQvHrleY6MG_AIxWw

althafvly, 2022-11-15:
It does support android 11. What is your issue in tool? send any logs from terminal

althafvly, 2022-11-15:
The attatched firmware seems to be unpacking fine.

```
althafvly@althafvly:~/Desktop/amlogic_kitchen$ ./unpack.sh
.....................
Amlogic Kitchen
.....................
.....................
Select level 1,2,3 or q/Q to exit: 
1
Deleting existing level1
.....................
Amlogic Kitchen
.....................
Files in input dir (*.img)
1 - ah212.HK1RBOX-X4_1000M_11.2022.03.05.13.49
.....................
Enter a file name :
ah212.HK1RBOX-X4_1000M_11.2022.03.05.13.49
[Msg]Image package version 0x2
[Msg]Unpack item [USB         ,              DDR] to (level1/DDR.USB) size:3248128 bytes
[Msg]Backup item [USB         ,            UBOOT] backItemId[0][USB, DDR]
[Msg]Unpack item [PARTITION   ,         _aml_dtb] to (level1/_aml_dtb.PARTITION) size:258048 bytes
[Msg]Unpack item [UBOOT       ,     aml_sdc_burn] to (level1/aml_sdc_burn.UBOOT) size:3248640 bytes
[Msg]Unpack item [ini         ,     aml_sdc_burn] to (level1/aml_sdc_burn.ini) size:602 bytes
[Msg]Unpack item [PARTITION   ,           boot_a] to (level1/boot_a.PARTITION) size:67108864 bytes
[Msg]Unpack item [PARTITION   ,       bootloader] to (level1/bootloader.PARTITION) size:3248128 bytes
[Msg]Unpack item [PARTITION   ,           dtbo_a] to (level1/dtbo_a.PARTITION) size:2097152 bytes
[Msg]Unpack item [PARTITION   ,             logo] to (level1/logo.PARTITION) size:910176 bytes
[Msg]Backup item [dtb         ,           meson1] backItemId[2][PARTITION, _aml_dtb]
[Msg]Unpack item [PARTITION   ,        odm_ext_a] to (level1/odm_ext_a.PARTITION) size:2588748 bytes
[Msg]Unpack item [conf        ,         platform] to (level1/platform.conf) size:29 bytes
[Msg]Unpack item [PARTITION   ,            super] to (level1/super.PARTITION) size:2030536568 bytes
[Msg]Unpack item [aml         ,         usb_flow] to (level1/usb_flow.aml) size:189008 bytes
[Msg]Unpack item [PARTITION   ,         vbmeta_a] to (level1/vbmeta_a.PARTITION) size:8192 bytes
[Msg]Unpack item [PARTITION   ,  vbmeta_system_a] to (level1/vbmeta_system_a.PARTITION) size:4096 bytes
[Msg]Unpack item [PARTITION   ,    vendor_boot_a] to (level1/vendor_boot_a.PARTITION) size:25165824 bytes
[Msg]Write config file "level1/image.cfg" OK!
Image unpack OK!
Done.
.....................
Amlogic Kitchen
.....................
.....................
Select level 1,2,3 or q/Q to exit: 
2
Deleting existing level2
Extracting odm_ext_a
.....Extraction from odm_ext_a.raw.img to odm_ext_a
.....Done! All extraction in odm_ext_a
.....Extraction from system_a.img to system_a
.....Done! All extraction in system_a
.....Extraction from system_ext_a.img to system_ext_a
.....Done! All extraction in system_ext_a
.....Extraction from vendor_a.img to vendor_a
.....Done! All extraction in vendor_a
.....Extraction from product_a.img to product_a
.....Done! All extraction in product_a
.....Extraction from odm_a.img to odm_a
.....Done! All extraction in odm_a
Done.
.....................
Amlogic Kitchen
.....................
.....................
Select level 1,2,3 or q/Q to exit: 
q
althafvly@althafvly:~/Desktop/amlogic_kitchen$ 

```

closed 2022-12-01 by althafvly