althafvly/AmlogicKitchen #11: pack_zip_to_aml "DDR.USB is missing, copy DDR.USB to level1 dir"  And create workflow xml file  failed

zengfr, 2022-10-07:
Python 3.9.13
.....................
Amlogic Kitchen
.....................
Files in input dir (*.zip)
1 - B863AV3.2M-220821_fota
.....................
Enter a file name :B863AV3.2M-220821_fota

7-Zip (a) 19.00 (x86) : Copyright (c) 1999-2018 Igor Pavlov : 2019-02-21

Scanning the drive for archives:
1 file, 554403244 bytes (529 MiB)

Extracting archive: in\B863AV3.2M-220821_fota.zip
--
Path = in\B863AV3.2M-220821_fota.zip
Type = zip
Physical Size = 554403244
Comment = signed by SignApk

Everything is Ok

Files: 25
Size:       560904600
Compressed: 554403244
sdat2img binary - version: 1.2

Android Nougat 7.x / Oreo 8.x detected!

Skipping command erase...
Copying 1 blocks into position 0...
Skipping command zero...
Copying 1 blocks into position 1...
Copying 4 blocks into position 9...
Copying 111 blocks into position 75...
Skipping command zero...
Done! Output image: F:\AmlogicKitchen-master\tmp\odm.img
已复制         1 个文件。
sdat2img binary - version: 1.2

Android Nougat 7.x / Oreo 8.x detected!

Skipping command erase...
Copying 1 blocks into position 0...
Copying 365 blocks into position 15374...
Copying 512 blocks into position 14862...
Copying 427 blocks into position 14346...
Copying 85 blocks into position 14777...
Copying 512 blocks into position 13834...
Copying 512 blocks into position 13322...
Copying 512 blocks into position 12810...
Copying 512 blocks into position 12298...
Copying 512 blocks into position 11786...
Copying 512 blocks into position 11274...
Copying 512 blocks into position 10762...
Copying 512 blocks into position 10250...
Copying 512 blocks into position 9738...
Copying 512 blocks into position 9226...
Copying 512 blocks into position 8714...
Copying 512 blocks into position 8202...
Copying 512 blocks into position 7690...
Copying 512 blocks into position 7178...
Copying 512 blocks into position 6666...
Copying 512 blocks into position 6154...
Copying 512 blocks into position 5642...
Copying 512 blocks into position 4618...
Copying 1 blocks into position 1...
Copying 4 blocks into position 9...
Copying 507 blocks into position 523...
[... 753 lines cut ...]
请按任意键继续. . .
"aml_sdc_burn.UBOOT is missing, copy aml_sdc_burn.UBOOT to level1 dir"
请按任意键继续. . .
"meson1.PARTITION is missing, copy meson1.PARTITION to level1 dir"
请按任意键继续. . .
"platform.conf is missing, copy platform.conf to level1 dir"
请按任意键继续. . .
[Msg]Pack Item[PARTITION   ,         _aml_dtb] from (level1\_aml_dtb.PARTITION),sz[0xfbae]B,
[Msg]Pack Item[ini         ,     aml_sdc_burn] from (level1\aml_sdc_burn.ini),sz[0x18a]B,
[Msg]Pack Item[PARTITION   ,             boot] from (level1\boot.PARTITION),sz[0xd10800]B,ft[normal]
[Msg]Pack Item[PARTITION   ,       bootloader] from (level1\bootloader.PARTITION),sz[0x14e970]B,ft[normal]
[Msg]Pack Item[PARTITION   ,             dtbo] from (level1\dtbo.PARTITION),sz[0x1b6]B,
[Msg]Pack Item[PARTITION   ,             logo] from (level1\logo.PARTITION),sz[0xfe080]B,ft[normal]
[Msg]Pack Item[PARTITION   ,              odm] from (level1\odm.PARTITION),sz[0x75070]B,ft[sparse]
[Msg]Pack Item[PARTITION   ,          product] from (level1\product.PARTITION),sz[0x3b6e0e0]B,ft[sparse]
[Msg]Pack Item[PARTITION   ,         recovery] from (level1\recovery.PARTITION),sz[0x1415800]B,ft[normal]
[Msg]Pack Item[PARTITION   ,           system] from (level1\system.PARTITION),sz[0x2db02bec]B,ft[sparse]
[Msg]Pack Item[PARTITION   ,           vbmeta] from (level1\vbmeta.PARTITION),sz[0x40]B,
[Msg]Pack Item[PARTITION   ,           vendor] from (level1\vendor.PARTITION),sz[0xf3db880]B,ft[sparse]
[Msg]version:0x2 crc:0x10e31abe size:1122261536 bytes[1070MB]
Pack image[out\B863AV3.2M-220821_fota.img] OK
Done.
Done.
请按任意键继续. . .
[image]

althafvly, 2022-10-07:
Did you copy the missing files?. You can't use that image without those files.

zengfr, 2022-10-07:
> Did you copy the missing files?. You can't use that image without those files.
1  origina img file is ok .
2  unpack origina file and  edit  
3  pack_level1_fota
4  pack_zip_to_aml  
5  out  img  is not ok .

althafvly, 2022-11-15:
```
"DDR.USB is missing, copy DDR.USB to level1 dir"
"UBOOT.USB is missing, copy UBOOT.USB to level1 dir"
"aml_sdc_burn.UBOOT is missing, copy aml_sdc_burn.UBOOT to level1 dir"
"meson1.PARTITION is missing, copy meson1.PARTITION to level1 dir"
"platform.conf is missing, copy platform.conf to level1 dir"
```

You didn't copy the missing files

```
[Msg]Pack Item[PARTITION , _aml_dtb] from (level1_aml_dtb.PARTITION),sz[0xfbae]B,
[Msg]Pack Item[ini , aml_sdc_burn] from (level1\aml_sdc_burn.ini),sz[0x18a]B,
[Msg]Pack Item[PARTITION , boot] from (level1\boot.PARTITION),sz[0xd10800]B,ft[normal]
[Msg]Pack Item[PARTITION , bootloader] from (level1\bootloader.PARTITION),sz[0x14e970]B,ft[normal]
[Msg]Pack Item[PARTITION , dtbo] from (level1\dtbo.PARTITION),sz[0x1b6]B,
[Msg]Pack Item[PARTITION , logo] from (level1\logo.PARTITION),sz[0xfe080]B,ft[normal]
[Msg]Pack Item[PARTITION , odm] from (level1\odm.PARTITION),sz[0x75070]B,ft[sparse]
[Msg]Pack Item[PARTITION , product] from (level1\product.PARTITION),sz[0x3b6e0e0]B,ft[sparse]
[Msg]Pack Item[PARTITION , recovery] from (level1\recovery.PARTITION),sz[0x1415800]B,ft[normal]
[Msg]Pack Item[PARTITION , system] from (level1\system.PARTITION),sz[0x2db02bec]B,ft[sparse]
[Msg]Pack Item[PARTITION , vbmeta] from (level1\vbmeta.PARTITION),sz[0x40]B,
[Msg]Pack Item[PARTITION , vendor] from (level1\vendor.PARTITION),sz[0xf3db880]B,ft[sparse]
[Msg]version:0x2 crc:0x10e31abe size:1122261536 bytes[1070MB]
Pack image[out\B863AV3.2M-220821_fota.img] OK
```

closed 2022-11-15 by althafvly