althafvly/AmlogicKitchen #13: Unpacking level 1 error

jayavanth, 2022-11-23:
Hi I'm trying to unpack Level 1 for a S905W2 ROM which has Android 11 and I'm getting this error

```
./unpack.sh 
.....................
Amlogic Kitchen
.....................
.....................
Select level 1,2,3 or q/Q to exit: 
1
Deleting existing level1
.....................
Amlogic Kitchen
.....................
No files found in /in
```

How can I fix it? Thanks!

althafvly, 2022-11-24:
`No files found in /in`

jayavanth, 2022-11-28:
> `No files found in /in`

Oh I see. So should I put the firmware in `/in`?

jayavanth, 2022-11-28:
Thanks! I'll try it out. I need to compile the binary for ARM

althafvly, 2022-12-01:
There won't support for arm. Some of the binaries used here are not opensource.

closed 2022-12-01 by althafvly