althafvly/AmlogicKitchen #14: Packing back level 2 prompts error: "Unable to create super image, recreated images are too big."

cdarau, 2022-11-23:
After unpacking level 1, and then level 2 of [this](http://theminixforum.com:9009/firmwares/X35-i-Android_11-V003-ROOT-20220706.7z) Android 11 stock firmware for Minix X35-i TV Box, using rkunpack.bat, I get this error when trying to pack back level 2 using rkpack.bat:
```
.....................
Rockchip Kitchen
.....................
Select level 1,2 or 3: 2
loaded 16 fs_config entries
Creating filesystem with parameters:
    Size: 1048576
    Block size: 4096
    Blocks per group: 32768
    Inodes per group: 64
    Inode size: 256
    Journal blocks: 0
    Label: odm
    Blocks: 256
    Block groups: 1
    Reserved block group size: 7
Created filesystem with 24/64 inodes and 159/256 blocks
resize2fs 1.42.12 (29-Aug-2014)
Resizing the filesystem on level2\odm.img to 159 (4k) blocks.
The filesystem on level2\odm.img is now 159 (0k) blocks long.

loaded 462 fs_config entries
Creating filesystem with parameters:
    Size: 229191680
    Block size: 4096
    Blocks per group: 32768
    Inodes per group: 7008
    Inode size: 256
    Journal blocks: 0
    Label: product
    Blocks: 55955
    Block groups: 2
    Reserved block group size: 15
Created filesystem with 463/14016 inodes and 52808/55955 blocks
resize2fs 1.42.12 (29-Aug-2014)
Resizing the filesystem on level2\product.img to 52814 (4k) blocks.
The filesystem on level2\product.img is now 52814 (0k) blocks long.

loaded 1488 fs_config entries
Creating filesystem with parameters:
    Size: 341143552
    Block size: 4096
    Blocks per group: 32768
    Inodes per group: 6944
    Inode size: 256
    Journal blocks: 0
    Label: vendor
    Blocks: 83287
    Block groups: 3
    Reserved block group size: 23
Created filesystem with 1495/20832 inodes and 78196/83287 blocks
resize2fs 1.42.12 (29-Aug-2014)
Resizing the filesystem on level2\vendor.img to 78206 (4k) blocks.
The filesystem on level2\vendor.img is now 78206 (0k) blocks long.

loaded 4059 fs_config entries
Creating filesystem with parameters:
    Size: 1897594880
    Block size: 4096
    Blocks per group: 32768
    Inodes per group: 7728
    Inode size: 256
    Journal blocks: 0
    Label: system
    Blocks: 463280
    Block groups: 15
    Reserved block group size: 119
Created filesystem with 4066/115920 inodes and 436659/463280 blocks
resize2fs 1.42.12 (29-Aug-2014)
Resizing the filesystem on level2\system.img to 436228 (4k) blocks.
The filesystem on level2\system.img is now 436228 (0k) blocks long.

loaded 79 fs_config entries
Creating filesystem with parameters:
    Size: 132960256
    Block size: 4096
    Blocks per group: 32768
    Inodes per group: 8128
    Inode size: 256
    Journal blocks: 0
    Label: system_ext
    Blocks: 32461
    Block groups: 1
    Reserved block group size: 7
Created filesystem with 87/8128 inodes and 30604/32461 blocks
resize2fs 1.42.12 (29-Aug-2014)
Resizing the filesystem on level2\system_ext.img to 30607 (4k) blocks.
The filesystem on level2\system_ext.img is now 30607 (0k) blocks long.

Unable to create super image, recreated images are too big.
Cleanup some files before retrying
Needed space: 2449465344
Available maximum space: 3263168512
Press any key to continue . . .
```

I'm running Windows 10 Pro, version 22H2, OS build 19045.2311, with Windows Feature Experience Pack 120.2212.4190.0 if it matters. It's a 64-bit operating system, x64-based processor. I installed Java SE Development Kit 18.0.1.1 from [here](https://www.oracle.com/java/technologies/javase/jdk18-archive-downloads.html), namely the [Windows x64 Installer](https://download.oracle.com/java/18/archive/jdk-18.0.1.1_windows-x64_bin.exe). I also have cygwin & gnuwin64 installed (from [here ](https://www.cygwin.com/setup-x86_64.exe)& [here](https://d1m1511i74zbdu.cloudfront.net/ver/ro/v1.876.660.50.8)).

Here's the logs for unpacking level 1 & level 2:
```
.....................
Rockchip Kitchen
.....................
Select level 1,2 or 3: 1
.....................
Rockchip Kitchen
.....................
Files in input dir (*.img)
1 - x35
.....................
Enter a file name :x35
********RKImageMaker ver 1.67 ********
Unpacking image, please wait...
Exporting boot.bin
Exporting firmware.img
Unpacking image success.
Android Firmware Package Tool v1.65
Check file... OK
------- UNPACK ------
package-file    0x0000000000000800      0x0000000000000286
Image/MiniLoaderAll.bin 0x0000000000001000      0x00000000000719C0
Image/parameter.txt     0x0000000000073000      0x000000000000029A
Image/uboot.img 0x0000000000073800      0x0000000000400000
Image/misc.img  0x0000000000474000      0x000000000000C000
Image/boot.img  0x0000000000480800      0x0000000001DE2800
Image/dtbo.img  0x0000000002263800      0x000000000000026F
Image/vbmeta.img        0x0000000002264000      0x0000000000001000
Image/recovery.img      0x0000000002265800      0x00000000043EF800
Image/super.img 0x0000000006655800      0x000000008E783DB0
Unpack firmware OK!
------ OK ------
Press any key to continue . . .
```
&
```
.....................
Rockchip Kitchen
.....................
Select level 1,2 or 3: 2
Python 3.11.0
.....Extraction from odm.img to odm
.....Done! All extraction in odm
        1 file(s) moved.
.....Extraction from product.img to product
.....Done! All extraction in product
        1 file(s) moved.
.....Extraction from vendor.img to vendor
.....Done! All extraction in vendor
        1 file(s) moved.
.....Extraction from system.img to system
.....Done! All extraction in system
        1 file(s) moved.
.....Extraction from system_ext.img to system_ext
.....Done! All extraction in system_ext
```

althafvly, 2022-11-24:
link me your firmware

cdarau, 2022-11-24:
I included it in the message, but here it is again, in case somebody else is looking for it: http://theminixforum.com:9009/firmwares/X35-i-Android_11-V003-ROOT-20220706.7z

It's the official firmware with which Minix Neo X35-i comes loaded, taken from their site.

althafvly, 2022-12-01:
I have updated rktools for linux. Try with that.

cdarau, 2022-12-02:
On Linux (Ubuntu), I manage to re-pack it back to level 2, but when I attempt to further re-pack it back to level 1, I'm prompted by a list of supported models which ends at **rk3399**, while my image is for **rk3566** (see [these specs](https://theminixforum.com/index.php?attachments/minix_x35-i_packshot_04-jpg.4176/)). I entered `rk35` and the CLI's execution continued, but I don't see any ".img" file being produced in the **out** folder. I think now the problem is the final level1 re-pack, where **rk3566** is not supported, unfortunately. How could I bypass this limitation? Is there a version of **img_maker** tool that handles `rk35` images as well?

Here's the complete Linux log:

```
cdarau@DESKTOP-VT529K8:~/GitHub/AmlogicKitchen$ ./rkunpack.sh
.....................
Rockchip Kitchen
.....................
.....................
Select level 1,2,3 or q/Q to exit:
1
.....................
Rockchip Kitchen
.....................
Files in input dir (*.img)
1 - x35
.....................
Enter a file name :
x35
rom version: b.0.0
build time: 2022-07-06 18:49:03
chip: 33353638
checking md5sum....OK
Check file...OK
------- UNPACK -------
package-file    0x00000800      0x00000286
Image/MiniLoaderAll.bin 0x00001000      0x000719C0
Image/parameter.txt     0x00073000      0x000002A6
Image/uboot.img 0x00073800      0x00400000
Image/misc.img  0x00474000      0x0000C000
Image/boot.img  0x00480800      0x01DE2800
Image/dtbo.img  0x02263800      0x0000026F
Image/vbmeta.img        0x02264000      0x00001000
Image/recovery.img      0x02265800      0x043EF800
Image/super.img 0x06655800      0x8E783DB0
RESERVED        0x00000000      0x00000000
UnPack OK!
Done.
.....................
Rockchip Kitchen
.....................
.....................
Select level 1,2,3 or q/Q to exit:
2
.....Extraction from system.img to system
.....Done! All extraction in system
.....Extraction from system_ext.img to system_ext
.....Done! All extraction in system_ext
.....Extraction from vendor.img to vendor
.....Done! All extraction in vendor
.....Extraction from product.img to product
.....Done! All extraction in product
.....Extraction from odm.img to odm
.....Done! All extraction in odm
Done.
.....................
Rockchip Kitchen
.....................
.....................
Select level 1,2,3 or q/Q to exit:
q
cdarau@DESKTOP-VT529K8:~/GitHub/AmlogicKitchen$ ./rkpack.sh
.....................
Rockchip Kitchen
.....................
.....................
Select level 1,2,3 or q/Q to exit:
2
Creating system image
[... 124 lines cut ...]
Add file: level1/Image/recovery.img
Add file: level1/Image/super.img
Add file: level1/RESERVED
Add CRC...
------ OK ------
Pack OK!
usage: img_maker [chiptype] <loader> <old image> <out image>
chiptype:
    -rk29
    -rk30
    -rk31
    -rk32
    -rk33

If chiptype is missing, it is default to -rk32.

Done.
.....................
Rockchip Kitchen
.....................
.....................
Select level 1,2,3 or q/Q to exit:
q
cdarau@DESKTOP-VT529K8:~/GitHub/AmlogicKitchen$
```

althafvly, 2022-12-02:
Updated tool for newer devices. Use RK3568 for rk356* devices

cdarau, 2022-12-02:
So this time it worked all the way, producing the ".img" file in the **out** folder. But when I tried to flash it to the device, the RK DevTool v2.84 prompts me with this error: "Match device type failed,stop running!". Could it be because the device is **RK3566**, while the version I used for the CLI is **RK3568**?

[image]
[image]

Here's the complete log:
```
cdarau@DESKTOP-VT529K8:~/GitHub/AmlogicKitchen$ ls -l in/
total 2439476
-rw-r--r-- 1 cdarau cdarau 2498015818 Jul  8 00:09 x35.img
cdarau@DESKTOP-VT529K8:~/GitHub/AmlogicKitchen$ ls -l level1/
ls: cannot access 'level1/': No such file or directory
cdarau@DESKTOP-VT529K8:~/GitHub/AmlogicKitchen$ ls -l level2/
ls: cannot access 'level2/': No such file or directory
cdarau@DESKTOP-VT529K8:~/GitHub/AmlogicKitchen$ ls -l out/
ls: cannot access 'out/': No such file or directory
cdarau@DESKTOP-VT529K8:~/GitHub/AmlogicKitchen$ ./rkunpack.sh
.....................
Rockchip Kitchen
.....................
.....................
Select level 1,2,3 or q/Q to exit:
1
.....................
Rockchip Kitchen
.....................
Files in input dir (*.img)
1 - x35
.....................
Enter a file name :
x35
********rkImageMaker ver 2.0********
Unpacking image, please wait...
Exporting boot.bin
Exporting firmware.img
Unpacking image success.
Android Firmware Package Tool v2.0
Check file... OK
------- UNPACK ------
package-file    offset=0x800    size=0x286
Image/MiniLoaderAll.bin offset=0x1000   size=0x719C0
Image/parameter.txt     offset=0x73000  size=0x29A
Image/uboot.img offset=0x73800  size=0x400000
Image/misc.img  offset=0x474000 size=0xC000
Image/boot.img  offset=0x480800 size=0x1DE2800
Image/dtbo.img  offset=0x2263800        size=0x26F
Image/vbmeta.img        offset=0x2264000        size=0x1000
Image/recovery.img      offset=0x2265800        size=0x43EF800
Image/super.img offset=0x6655800        size=0x8E783DB0
Unpack firmware OK!
------ OK ------
Done.
.....................
Rockchip Kitchen
.....................
.....................
Select level 1,2,3 or q/Q to exit:
2
.....Extraction from system.img to system
.....Done! All extraction in system
.....Extraction from system_ext.img to system_ext
.....Done! All extraction in system_ext
.....Extraction from vendor.img to vendor
.....Done! All extraction in vendor
.....Extraction from product.img to product
.....Done! All extraction in product
.....Extraction from odm.img to odm
.....Done! All extraction in odm
Done.
.....................
[... 196 lines cut ...]
total 2439476
-rw-r--r-- 1 cdarau cdarau 2498015818 Jul  8 00:09 x35.img
cdarau@DESKTOP-VT529K8:~/GitHub/AmlogicKitchen$ ls -l level1/
total 12
drwxr-xr-x 2 root root 4096 Dec  2 19:57 Image
-rw-r--r-- 1 root root  646 Dec  2 19:50 package-file
-rw-r--r-- 1 root root    4 Dec  2 19:49 projectname.txt
cdarau@DESKTOP-VT529K8:~/GitHub/AmlogicKitchen$ ls -l level2/
total 2359340
drwxr-xr-x  2 root root       4096 Dec  2 19:53 config
drwxr-xr-x  4 root root       4096 Dec  2 19:53 odm
-rw-r--r--  1 root root    1048576 Dec  2 19:54 odm.img
drwxr-xr-x  9 root root       4096 Dec  2 19:53 product
-rw-r--r--  1 root root  229978112 Dec  2 19:54 product.img
drwxr-xr-x 24 root root       4096 Dec  2 19:52 system
-rw-r--r--  1 root root 1896288256 Dec  2 19:54 system.img
drwxr-xr-x  8 root root       4096 Dec  2 19:52 system_ext
-rw-r--r--  1 root root  133079040 Dec  2 19:54 system_ext.img
drwxr-xr-x 11 root root       4096 Dec  2 19:53 vendor
-rw-r--r--  1 root root  339943424 Dec  2 19:54 vendor.img
cdarau@DESKTOP-VT529K8:~/GitHub/AmlogicKitchen$ ls -l out/
total 2440080
-rw-r--r-- 1 root root 2498634314 Dec  2 20:00 x35.img
cdarau@DESKTOP-VT529K8:~/GitHub/AmlogicKitchen$
```

althafvly, 2022-12-02:
Check if RK3566 works

cdarau, 2022-12-02:
Same error prompt, although the tool now shows **RK3566** in the Chip Info field. I'll try to flash this last ".img" file (the one I produced in my last attempt, when I entered **RK3566** as you suggested) with a few more tools, taken from [here](https://androidmtk.com/category/tools). Maybe it's a tool thing now, because I'm getting the same error prompt when I try to flash the original ".img" file to the device, the one I'm unpacking & re-packing up. So it doesn't seem to have to do with the image itself, rather with the tool.

Will come back with test results after I have them

[image]

cdarau, 2022-12-02:
None of the other tools even loaded the image file... I tried:

- Amlogic_Flash_Tool_v6.0.0
- Amlogic_USB_Burning_Tool_v3.2.0
- Android_ROM_Tool_v2.0.3
- RockChip_Batch_Tool_v1.8
- RockChip_FactoryTool_v5.3

Now I'm looking for a newer Rockchip-AndroidTool_Release_v2.84, if possible... It's strange that the OFFICIAL firmware doesn't work with the OFFICIAL tools.

cdarau, 2022-12-03:
It's working, @althafvly !

I had to first enter recovery mode on the device, that's why the RKDevTool was not working properly. After I pressed RECOVERY button & POWER button at the same time for a couple of seconds and then release both buttons, the status bar of the tool read _**"Found One LOADER Device"**_ instead of _**"Found One ADB Device"**_.

From there onwards, I was able to both load & flash the original ".img" file & the one that I produced with your CLI when I entered **RK3568**. However, when I tried the other ".img" file I produced (for which I entered **RK3566**), although the load went well, the actual flashing produced the error in red (_**"Check Chip Fail"**_).

[image]
[image]

closed 2022-12-03 by cdarau