althafvly/AmlogicKitchen #18: rkpack level 1 error

kalehrl, 2023-04-15:
I get the following error when I try to pack a Rockchip image level 1:
```
root@debian:/home/kalehrl/Templates/android/AmlogicKitchen# ./rkpack.sh
.....................
Rockchip Kitchen
.....................
.....................
Select level 1,2,3 or q/Q to exit: 
1
Supported models:
px30            RKPX30
px3se           RK312A
rk1808          RK180A
rk3036          RK303A
rk3128h         RK312X
rk312*          RK312A
rk322*          RK322A
rk3229          RK3229
rk3288          RK320A
rk3308          RK3308
rk3326          RK3326
rk3328          RK322H
rk3368          RK330A
rk3399*         RK330C
rk356*          RK3568
rv1126_rv1109   RK1126
If your model starts with rk312* then its RK312A
Enter your chip model, Eg: RK312X:
RK3229
Android Firmware Package Tool v2.0
------ PACKAGE ------
Error:<PackImage> uboot was overlapped in the parameter!
------ FAILED ------                                                                                          
********rkImageMaker ver 2.0********                                                                          
Generating new image, please wait...                                                                          
Error:Get image version failed!                                                                               
Done.
```

althafvly, 2023-08-20:
Share link to your firmware. I can check

kalehrl, 2023-08-23:
https://www.filefactory.com/file/2pn3aa7eyp99/%28mxq-4k%29_1G8G_rtl8189_20171113_r1.rar

althafvly, 2023-08-23:
Trying using RK322A

closed 2023-12-08 by althafvly