althafvly/AmlogicKitchen #26: Ferramenta tv box

Thiagopb35, 2024-12-24:
I managed to make all the flash tools work on Linux Ubuntu 24.04.1 LTS but now I'm looking for tools to record Mac and serial on the Amlogic Rockchip and Allwinner boxes as well as a tool to dump the firmware on the Amlogic Rockchip and Allwinner boxes in mode upgrade without the box needing to be rooted and remove all partitions including aml_dtb Do you know if it exists or if there is a way to do it preferably in .sh

althafvly, 2024-12-28:
None that I know of.

closed 2024-12-28 by althafvly

Thiagopb35, 2024-12-28:
but you are a genius my friend developed the aml flash tool there is the amlogic kitchen livesuit among others make one to record the Mac adrees and a serial number of up to 32 digits so you should be drinking candy from a child.... I like it work with Linux Ubuntu but I have these pending Mac and serial tools and one that dumps the firmware of the Allwinner Amlogic and Rockchip boxes in upgrade mode that does not require root with support up to Android 11