althafvly/AmlogicKitchen #30: Unable to perform level2 amlpacking

lhplab, 2025-03-08:
Hello

First : let me thank you for your work!

Here is the issue I'm facing:
When trying to perform a level2 repack, I've got this issue (on a French System):

```
Select level 1,2,3 or q/Q to exit: 
2
Creating system image
./amlpack.sh : ligne 34 :  6007 Instruction non permise bin/make_ext4fs -s -J -L $part -T -1 -S $fc -C $fs -l $size -a $part level1/$part.PARTITION level2/$part
```

Not quite sure of the translation but should be something like this on an English system :

`./amlpack.sh : line 34 :  6007 Instruction not allowed bin/make_ext4fs -s -J -L $part -T -1 -S $fc -C $fs -l $size -a $part level1/$part.PARTITION level2/$part
`

It seems maybe that make_ext4fs is deprecated and should be replaced by "mke2fs" or "mkfs.ext4" if I refer to the following link : [https://askubuntu.com/questions/1448624/make-ext4fs-command-not-found-in-ubuntu-22-04/1481982#1481982](url) ?

But I'm not really sure...

Thanks for your help!
Regards
Lhplab

althafvly, 2025-03-08:
Share link to the firmware

lhplab, 2025-03-08:
This is a ROM foud at [https://slimboxtv.ru/x96mini/](url) :
 file sbx_x96mini_atv_9_16.zip retreived from this page :
[https://disk.yandex.ru/d/jbVMNMjVUzdxmQ](url)

I've removed some apks and I am trying to repack it to a new image. I will give it a try "as is" without any modification and I will tell you if I still get the message

Thanks

lhplab, 2025-03-08:
I did try from the original ROM performing:

./amlunpack.sh
=> level1
=> then level2
=> then level3

Nothing modified on the files unpacked from the ROM then

./amlpack.sh
=> level3
=> leve2

And I get the following error :

```
.....................
.....................
Select level 1,2,3 or q/Q to exit: 
2
Creating system image
./amlpack.sh : ligne 34 :  7085 Instruction non permise bin/make_ext4fs -s -J -L $part -T -1 -S $fc -C $fs -l $size -a $part level1/$part.PARTITION level2/$part
.....................

```

Here is my system configuration 
```
xxx@xxxxxxxx:~/X96/Tools/AmlogicKitchen$ uname -a
xxx@xxxxxxxx:~/X96/Tools/AmlogicKitchen$ Linux xxxxxxxx 5.10.0-33-amd64 #1 SMP Debian 5.10.226-1 (2024-10-03) x86_64 GNU/Linux

```

lhplab, 2025-03-08:
I'va made a little modification to the script to be able to see the parameters value (lines 40 to 43):

```
        if [ ! -z "$size" ]; then
          echo bin/make_ext4fs -s -J -L $part -T -1 -S $fc -C $fs -l $size -a $part level1/$part.PARTITION level2/$part
          bin/make_ext4fs -s -J -L $part -T -1 -S $fc -C $fs -l $size -a $part level1/$part.PARTITION level2/$part
        fi

```

Here is the output of the command now:

```
Select level 1,2,3 or q/Q to exit: 
2
Creating system image
bin/make_ext4fs -s -J -L system -T -1 -S level2/config/system_file_contexts -C level2/config/system_fs_config -l 1342177280 -a system level1/system.PARTITION level2/system
./amlpack.sh : ligne 34 :  7262 Instruction non permise bin/make_ext4fs -s -J -L $part -T -1 -S $fc -C $fs -l $size -a $part level1/$part.PARTITION level2/$part

```

lhplab, 2025-03-08:
Even without using your script I can't launch the make_ext4fs binary:

```
xxx@xxxxxxxx:~/X96/Tools/AmlogicKitchen$ sudo bin/make_ext4fs 
Instruction non permise

```

althafvly, 2025-03-08:
Which distro and `arch` are you running this tool?

lhplab, 2025-03-08:
Debian 11 (Bullseye) kernel 5.10 x86_64

althafvly, 2025-03-08:
Use ubuntu 22, as these built for it.

lhplab, 2025-03-08:
Thank you very much for your help!

Thanks to your comment (make_ext4fs was compiled for Ubuntu) I've try to compile it on my computer using the source files from this repository https://github.com/iglunix/make_ext4fs and it seems to work! :-)

Thank you very much and have a nice day! :-)

closed 2025-03-08 by lhplab

lhplab, 2025-03-08:
I finally had to recompile another version of make_ext4fs (https://github.com/superr/make_ext4fs/tree/master) because the ROM that I got with https://github.com/iglunix/make_ext4fs did not boot