althafvly/AmlogicKitchen #34: AllWinner - bin/imgrepacker: cannot execute: required file not found

tv4you2016, 2025-03-21:
Hi, I'm having this error when decompiling for AllWinner

```
root@RK3576:~/AmlogicKitchen# ./awunpack.sh
.....................
AllWinner Kitchen
.......................
...................
Select level 1,2,3 or q/Q to exit:

1
....................
AllWinner Kitchen
.....................
./awunpack.sh: line 30: rename: command not found
Files in input directory (*.img)
1 - X96QProPlus_20250218-1427-002
....................
Enter a filename:
X96QProPlus_20250218-1427-002
./awunpack.sh: line 50: bin/imgrepacker: cannot execute: required file not found
Done.
.....................
AllWinner Kitchen
.....................
..................
Select level 1,2,3 or q/Q to exit:
```

dataghost2014, 2025-04-02:
Same here, not sure why. Do you have a variant of Java installed?

althafvly, 2025-04-03:
Use ubuntu 22.04 in a x86_64

dataghost2014, 2025-04-03:
Still not working with 22.04 wsl. Just tested, RK images and AML images working

althafvly, 2025-04-04:
Wsl is not supported

dataghost2014, 2025-04-04:
I used it last year, not sure why all the sudden it stopped working Ubuntu is Ubuntu, it should matter if it's WSL or not it's still the same Ubuntu platform I use it for all development

closed 2025-04-04 by althafvly