althafvly/AmlogicKitchen #36: Update on awunpack and awpack

dataghost2014, 2025-04-02:
Both will not execute, same error on both, after setting permissions no change. Not sure why, both are the same error.

./imgpack: cannot execute: required file not found
./imgrepacker: cannot execute: required file not found

the other binaries are opening, Ill see if anything is missing on my end, thanks for the support, Cheers!

HighwayStar, 2025-04-23:
facing same issue on fedora 42, it was working on older system

HighwayStar, 2025-04-23:
Seems some problem with 32 bit binary on 64 bit system.

GCB2023, 2025-05-15:
the old version of long long ago is work on same system, but new version not work show error message "bin/imgpack: No such file or directory", the system is ubuntu22.04

althafvly, 2025-05-16:
> the old version of long long ago is work on same system, but new version not work show error message "bin/imgpack: No such file or directory", the system is ubuntu22.04

imgpack and imgrepacker have not changed since it was added. It should work.

Try

sudo dpkg --add-architecture i386
sudo apt update

GCB2023, 2025-05-16:
> imgpack and imgrepacker have not changed since it was added. It should work.
>
> Try
>
> sudo dpkg --add-architecture i386 sudo apt update

We use two version in same time, same machine, same system, same firmware, did not change anything.  the old work, the new not work.

althafvly, 2025-05-16:
> We use two version in same time, same machine, same system, same firmware, did not change anything. the old work, the new not work.

what do you mean by old?.

GCB2023, 2025-05-16:
> what do you mean by old?.

can run in windows and linux, download on Dec 2023.

althafvly, 2025-05-16:
> can run in windows and linux, download on Dec 2023.

Tool is no longer maintained for windows. In linux you must enable 32bit for few binaries.

Check this.
https://discuss.getsol.us/d/3682-how-to-run-elf-32-bit/6

closed 2025-05-27 by althafvly