althafvly/AmlogicKitchen #43: Segmentation fault bin/extract.erofs during level 2 extraction with both ampack and aml_image_v2_packer

kalehrl, 2025-07-06:
I get this error:
```
Extracting system_a
Extract: Starting...
Extract: fs_config|file_contexts|fs_options  saving...
Extract: fs_config|file_contexts|fs_options  done.
Extract: [ 100.00% ]
./common/extract_images.sh: line 17:  1701 Segmentation fault      bin/extract.erofs -i "$IMAGE_FILE" -x -o "$OUTPUT_FOLDER"
```
during level 2 extraction with both ampack and aml_image_v2_packer.
The ROM is here:
https://disk.yandex.ru/d/DmZYHhpaxlBZGQ
Thank you.

kalehrl, 2025-07-06:
No. I use Debian virtual machine.

althafvly, 2025-07-06:
Might not work in virtual machine.

kalehrl, 2025-07-06:
Hmm, everything else works fine.

kalehrl, 2025-07-10:
I learned that this firmware is encrypted so that's probably the cause of the problems.

JasonBrooks1984, 2025-07-23:
This is an issue I am getting on anything ubuntu, debian, fedora, has not been resolved my friend. WSL gets the same issue as another virtual machine, or as a stand alone install, wiped my windows and still no dice, is there any resolve to this? Thanks